ros-drivers/openni_camera #31: camera_info_url empty when calib files at default location

ahoarau, 2015-01-20:
When I calibrate my kinects (using the ros calibrator.py), it is saving the files at :
```
~/.ros/camera_info/rgb_0000000000000000.yml 
~/.ros/camera_info/depth_0000000000000000.yml 
```
But the openni driver does **not** populates 
```
/camera/driver/rgb_camera_info_url
/camera/driver/depth_camera_info_url
```

> ```
$ rosparam get /camera/driver/rgb_camera_info_url 
$ ''
```
Even though rgb_info_manager_->isCalibrated() is **true**.

I opened a [pull request](https://github.com/ros-perception/image_common/pull/30) for the camera_info_manager, but meanwhile we can add this **patch** : 
```cpp
// The default location for calibration files
const std::string default_camera_info_url = "file://${ROS_HOME}/camera_info/${NAME}.yaml";
// Checking if rgb_info_url has been correctly populated : 
if(rgb_info_url.empty())
 {
     rgb_info_url = rgb_info_manager_->resolveURL(default_camera_info_url,rgb_name);
     param_nh.setParam("rgb_camera_info_url",rgb_info_url);
 }
if(ir_info_url.empty())
 {
     ir_info_url = ir_info_manager_->resolveURL(default_camera_info_url,ir_name);
     param_nh.setParam("depth_camera_info_url",ir_info_url);
 }
```

below :
 ```cpp
  std::string rgb_info_url, ir_info_url;
  param_nh.param("rgb_camera_info_url", rgb_info_url, std::string());
  param_nh.param("depth_camera_info_url", ir_info_url, std::string());
/////...
  // Load the saved calibrations, if they exist
  rgb_info_manager_ = boost::make_shared<camera_info_manager::CameraInfoManager>(rgb_nh, rgb_name, rgb_info_url);
  ir_info_manager_  = boost::make_shared<camera_info_manager::CameraInfoManager>(ir_nh,  ir_name,  ir_info_url);
  
  if (!rgb_info_manager_->isCalibrated())
    NODELET_WARN("Using default parameters for RGB camera calibration.");
  if (!ir_info_manager_->isCalibrated())
    NODELET_WARN("Using default parameters for IR camera calibration.");
 ```

And **then** :+1: 
```
rosparam get /camera/driver/rgb_camera_info_url 
file:///home/hoarau/.ros/camera_info/rgb_0000000000000000.yaml
```

jack-oquin, 2015-01-20:
It does not seem appropriate to me for the driver to set its own values in the parameter server. That is not normal ROS node behavior. 

If a parameter is null or not set, nodes will normally provide a default. But, changing the parameter value is not normally done at that level.

ahoarau, 2015-01-21:
I'm trying to work with the [python version](https://github.com/ros-perception/camera_info_manager_py) which cannot access to the camera's serial number (ot's not a driver), so it cannot get the default depth_${serial}.yml and and rgb_${serial}.yml.
```python
cm = CameraInfoManager()
cm.loadCameraInfo()
cm.getURL()
```
returns :
```
file:///home/hoarau/.ros/camera_info/camera.yaml
```
A way to get the loaded calibration file would be to use the parameter server and then use the url argument in the CameraInfoManager's constructor: 
```python
rgb_url = rospy.get_param(cname+'/driver/rgb_camera_info_url').replace('file://','')
ir_url = rospy.get_param(cname+'/driver/depth_camera_info_url').replace('file://','')
```

We could also add "get_camera_info" service (from c++) and pass it as a constructor for the python version. What do you think ?

jack-oquin, 2015-01-21:
If I understand the problem, you are trying to access the calibration information in a Python script. But, the default URL contains ${NAME} and the camera serial number is not known to that script, so it can't be resolved there. Is that right?

The reason why the URL is provided as a parameter is to allow a system integrator to store it in any convenient location. They default location was chosen to avoid multiple cameras interfering with each other. That is not convenient for you. It was chosen for people who do not care where the camera info goes.

Since you do care where the file is, why not put it in some convenient package or filesystem location by defining that parameter explicitly? Your script and the driver can both read the name from the parameter server. They would not be affected if you needed to replace the camera at some point, although that would cause the serial number to change. 

You can create a package containing all your camera info files, then reference it using something like:
```
 roslaunch openni_launch openni.launch rgb_camera_info_url:="package://your_cameras/info/kinect_rgb.yaml" depth_camera_info_url:="package://your_cameras/info/kinect_depth.yaml"
```

My objection to setting the parameter in the driver is that those names should be determined at a higher level in the system. You can take advantage of that by making the calibration file part of your system design.

ahoarau, 2015-01-22:
> If I understand the problem, you are trying to access the calibration information in a Python script. But, the default URL contains ${NAME} and the camera serial number is not known to that script, so it can't be resolved there. Is that right?

Precisely. 
> The reason why the URL is provided as a parameter is to allow a system integrator to store it in any convenient location. They default location was chosen to avoid multiple cameras interfering with [...]

The default location is perfectly fine as each camera has a unique serial number. 
> Since you do care where the file is, why not put it in some convenient package or filesystem location by defining that parameter explicitly? Your script and the driver can both read the name from [...]

I do not care where the file is (the default location is good), I just want my python script to find the one that is loaded by the driver so we can talk with the same calibration matrices and not be confused by URLs. 

My main problem is that I'm switching depth sensors quite often (old kinects, xtions etc). The driver finds the correct calibration (at default location->rgb_${serial}, and that's perfect !), so why not the camera_info_manager.py ?

>My objection to setting the parameter in the driver is that those names should be determined at a higher level in the system. You can take advantage of that by making the calibration file part of your system design.

Setting the parameter in the driver would just be for the case where you don't provide a calib file url. In that only case, set the parameter. Otherwise, it's the user's URL. So in any case, those URLs are never empty.

>note : I should have updated my previous comment like this :

```python
# I have the intrinsics of my_kinect at default location, i.e file:///home/hoarau/.ros/camera_info/depth_${my_kinect_serial}.yaml

# I launch the driver : 
openni_launch camera:=my_kinect

# Then in my script :
k1 = CameraInfoManager("my_kinect") # no url provided
k1.loadCameraInfo()
k1.getURL() # -> should returns file:///home/hoarau/.ros/camera_info/depth_${my_kinect_serial}.yaml

```

jack-oquin, 2015-01-22:
The driver *should not* set its own values in the parameter server. That is not normal ROS driver behavior. Doing that would be strange and surprising for everyone but you. You are free to fork the driver and patch your copy to behave that way, but I do not want to merge that patch into the standard driver.

I still do not understand why you cannot solve this problem by setting the URL in the normal way. What happens if you set it to the location you seem to prefer?
```
file:///home/hoarau/.ros/camera_info/rgb_0000000000000000.yaml
```
Why that does that not work?

ahoarau, 2015-01-22:
It works, but it should be done **automatically** by the manager.py. 
I'm convinced there should be a simpler way to get the camera's calibration data once the driver is launched.
A ***get_camera_info service*** would work.

jack-oquin, 2015-01-22:
Experienced ROS users often start ``roscore`` before launching applications, so they can leave it,  ``rviz`` and other utilities running for multiple tests. 

If the driver were patched as you suggest, a developer who swaps cameras between tests would be surprised to discover that the new camera is no longer accessible, because the driver had overwritten the empty URL with a string containing the actual serial number of the first camera. Other drivers do not act that way, and openni_camera should not, either. If it did, I would open a bug report to remove the feature you are proposing.

Since your problem can already be solved using the existing API, I respectfully suggest that you close this issue.

closed 2015-01-23 by ahoarau