ros-drivers/openni_camera #33: jade release?

furushchev, 2015-06-18:
no apt repository in jade release.
Is there any reason/diffs not to release openni related repositories apt?

furushchev, 2015-06-23:
ping

jonbinney, 2015-06-24:
I'm still on indigo; is there a libopenni-dev package available for ubuntu 15.04? That would need to be released before openni_camera.

jack-oquin, 2015-06-25:
Looks like it at least exists: https://launchpad.net/ubuntu/vivid/+package/libopenni-dev

fizzikz, 2015-07-21:
To install openni _camera/launch on vivid. 
1. First setup your catkin_ws folder, like in the beginner ROS tutorial http://wiki.ros.org/catkin/Tutorials/create_a_workspace
2. Then in your catkin_ws/src folder, git clone the stacks that you need.

3. run rosdep to see if you are missing any dependencies 
>rosdep update
>rosdep check --from-paths . --ignore-src --rosdistro jade

4. If you are run rosdep install to get all the missing dependencies.
> rosdep install --from-paths . --ignore-src --rosdistro jade -y

5. then go into your catkin_ws folder and run catkin_make
> cd ~/catkin_ws/https://github.com/avin2/SensorKinectwi
>catkin_make

6. Finally, go to https://github.com/avin2/SensorKinect and install the sensor driver following the linux directions. And it should work after that.

Linux:
 Requirements:
  1) GCC 4.x
     From: http://gcc.gnu.org/releases.html
     Or via apt:
     sudo apt-get install g++
  2) Python 2.6+/3.x
     From: http://www.python.org/download/
     Or via apt:
     sudo apt-get install python
  3) OpenNI 1.5.x.x
     From: http://www.openni.org/Downloads/OpenNIModules.aspx
     
 Building Sensor:
  1) Go into the directory: "Platform/Linux/CreateRedist".
     Run the script: "./RedistMaker".
     This will compile everything and create a redist package in the "Platform/Linux/Redist" directory.
     It will also create a distribution in the "Platform/Linux/CreateRedist/Final" directory.
  2) Go into the directory: "Platform/Linux/Redist".
     Run the script: "sudo ./install.sh" (needs to run as root)

       The install script copies key files to the following location:
         Libs into: /usr/lib
         Bins into: /usr/bin
         Config files into: /usr/etc/primesense
         USB rules into: /etc/udev/rules.d 
         Logs will be created in: /var/log/primesense
   
  To build the package manually, you can run "make" in the "Platform\Linux\Build" directory.
  
  Important: Please note that even though the directory is called Linux, you can also use it to compile it for 64-bit targets and pretty much any other linux based environment. 

 Building Sensor using a cross-compiler:
  1) Make sure to define two environment variables:
     - _CXX - the name of the cross g++ for platform 
     - _STAGING - a path to the staging dir (a directory which simulates the target root filesystem).
     Note that  should be upper cased.
     For example, if wanting to compile for ARM from a x86 machine, ARM_CXX and ARM_STAGING should be defined.
  2) Go into the directory: "Platform/Linux/CreateRedist".
     Run: "./RedistMaker " (for example: "./RedistMake Arm").
     This will compile everything and create a redist package in the "Platform/Linux/Redist" directory.
     It will also create a distribution in the "Platform/Linux/CreateRedist/Final" directory.
  3) To install OpenNI files on the target file system:
     Go into the directory: "Platform/Linux/Redist".
     Run the script: "./install.sh -c $_STAGING" (for example: "./install.sh -c $ARM_STAGING").

       The install script copies key files to the following location:
         Libs into: STAGING/usr/lib
         Bins into: STAGING/usr/bin
         Config files into: STAGING/usr/etc/primesense
         USB rules into: STAGING/etc/udev/rules.d 
         Logs will be created in: STAGING/var/log/primesense
   
  To build the package manually, you can run "make PLATFORM=" in the "Platform\Linux\Build" directory.
  If you wish to build the Mono wrappers, also run "make PLATFORM= mono_wrapper" and "make PLATFORM= mono_samples".
  

  Excerpted from README.md from https://github.com/avin2/SensorKinect.

furushchev, 2015-07-24:
Thanks all. I know how to install from source, just to know whether it will be released soon on jade, and if not, I'd like to clarify what problem is there.

jonbinney, 2015-07-24:
@furushchev the problem is that this package isn't really getting maintained/release by anyone. There's a few reasons, but the biggest are probably:
- this package was primarily for primesense devices, which aren't sold anymore because that company no longer exists
- most people have switched to openni2_camera (for primesense devices) or freenect (for kinects)
- the previous maintainers listed in the package.xml no longer have moved on to other jobs that don't involve this package

Options at this point:
- find a new maintainer if someone has time to commit and will be using this package for quite a while
- switch to openni2_camera or freenect
- use this package from source

furushchev, 2015-09-27:
@jonbinney I'm sorry for very late response, and thank you for clarifying of this problem. OK, I will try to use openni2_camera for a while.

closed 2015-09-27 by furushchev