ros-drivers/openni_camera #36: Parameters ~image_mode and ~depth_mode still work ? how can i use it

lligen, 2015-09-07:
hi,
I want the kinect to output 320x240 rgb and depth images directly, i tried to set this para in launch file but failed ,
**~image_mode** (int, default: 2)
Image output mode for the color/grayscale image Possible values are: SXGA_15Hz (1): 1280x1024@15Hz, VGA_30Hz (2): 640x480@30Hz, VGA_25Hz (3): 640x480@25Hz, QVGA_25Hz (4): 320x240@25Hz, QVGA_30Hz (5): 320x240@30Hz, QVGA_60Hz (6): 320x240@60Hz, QQVGA_25Hz (7): 160x120@25Hz, QQVGA_30Hz (8): 160x120@30Hz, QQVGA_60Hz (9): 160x120@60Hz
**~depth_mode** (int, default: 2)
depth output mode Possible values are: SXGA_15Hz (1): 1280x1024@15Hz, VGA_30Hz (2): 640x480@30Hz, VGA_25Hz (3): 640x480@25Hz, QVGA_25Hz (4): 320x240@25Hz, QVGA_30Hz (5): 320x240@30Hz, QVGA_60Hz (6): 320x240@60Hz, QQVGA_25Hz (7): 160x120@25Hz, QQVGA_30Hz (8): 160x120@30Hz, QQVGA_60Hz (9): 160x120@60Hz

Could you please tell me how to set this parameters?

Thank you vary much~~

jack-oquin, 2015-09-07:
This question seems more appropriate on http://answers.ros.org

130s, 2015-10-17:
Looks like asked and solved http://answers.ros.org/question/217252/how-to-make-openni_camera-package-to-output-320x240-rgb-and-depth-images-directly/

closed 2015-10-17 by 130s