ros-drivers/openni_camera #45: Release into ROS Kinect

130s, 2016-03-31:
As of today bloom failed. I don't know what to do to proceed from here. [Google search](https://www.google.com/search?client=ubuntu&channel=fs&q=fedora+24++libusbx-devel+&ie=utf-8&oe=utf-8) doesn't clearly tell me whether the library in question (`libusbx-devel` or similar on version Fedora 24) is already available. It's [missing from rosdep key](https://github.com/ros/rosdistro/blob/a06e8229776041fe9250f36596ad48907e8abb10/rosdep/base.yaml#L2518-L2524).
```
$ bloom-release --rosdistro kinetic --track kinetic openni_camera
:
libusb-1.0-dev is depended on by these packages: ['openni_camera']
<== Failed
Could not resolve rosdep key 'libusb-1.0-dev' for distro '24':
No definition of [libusb-1.0-dev] for OS version [24]
	rosdep key : libusb-1.0-dev
	OS name    : fedora
	OS version : 24
	Data: arch:
- libusbx
debian:
- libusb-1.0-0-dev
fedora:
  '21':
  - libusbx-devel
  '22':
  - libusbx-devel
  beefy:
  - libusb1-devel
  heisenbug:
  - libusbx-devel
  "schr\xF6dinger\u2019s":
  - libusbx-devel
  spherical:
  - libusbx-devel
gentoo:
- =dev-libs/libusb-1.0*
macports:
- libusb
opensuse:
- libusb-1_0-devel
ubuntu:
- libusb-1.0-0-dev

Failed to resolve libusb-1.0-dev on fedora:24 with: Error running generator: Failed to resolve rosdep key 'libusb-1.0-dev', aborting.
libusb-1.0-dev is depended on by these packages: ['openni_camera']
<== Failed
Some of the dependencies for packages in this repository could not be resolved by rosdep.
You can try to address the issues which appear above and try again if you wish.
```

jack-oquin, 2016-04-01:
Is this just a missing key, or do we not know how to set it?

@cottsay: can you suggest any pointers for finding corresponding Fedora 24 packages?

130s, 2016-04-02:
> Is this just a missing key, or do we not know how to set it?

The rosdep key for the necessary package for Fedora 24 is missing for sure. What I'm unsure about is if package is already available.
Yeah so input from @cottsay or anyone who's knowledgeable in Fedora is appreciated.

cottsay, 2016-04-02:
Looks like `libusbx` is still the right package in f23/f24. I haven't gotten around to adding keys for Kinetic Fedora releases yet.

https://admin.fedoraproject.org/pkgdb/package/rpms/libusbx/

To unblock you, add the same keys as f21/f22:
```
    '23': [libusbx-devel] 
    '24': [libusbx-devel] 
```

Thanks for looping me in instead of skipping Fedora generation.

--scott

130s, 2016-04-04:
Release requested https://github.com/ros/rosdistro/pull/10981
Thank you @jack-oquin @cottsay

closed 2016-04-04 by 130s