ros-drivers/openni_camera #54: 'process has died' while I got the warning 'USB events thread - failed to set priority.'

blueeaglex, 2017-07-11:
Hardware:1.kinect1.0 for xbox360
2.PC with usb2.0
Software: Ubuntu16.04 LTS, ros-kinetic-desktop-full
I tried to excute openni.launch and get these errors:
`auto-starting new master
process[master]: started with pid [9207]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8c588874-654d-11e7-ab6f-00e0b412588a
process[rosout-1]: started with pid [9220]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [9237]
process[camera/driver-3]: started with pid [9238]
process[camera/rgb_debayer-4]: started with pid [9239]
process[camera/rgb_rectify_mono-5]: started with pid [9245]
process[camera/rgb_rectify_color-6]: started with pid [9258]
process[camera/ir_rectify_ir-7]: started with pid [9267]
process[camera/depth_rectify_depth-8]: started with pid [9282]
[ INFO] [1499676940.858628822]: Initializing nodelet with 4 worker threads.
process[camera/depth_metric_rect-9]: started with pid [9297]
process[camera/depth_metric-10]: started with pid [9322]
process[camera/depth_points-11]: started with pid [9337]
process[camera/register_depth_rgb-12]: started with pid [9343]
process[camera/points_xyzrgb_sw_registered-13]: started with pid [9349]
process[camera/depth_registered_sw_metric_rect-14]: started with pid [9366]
process[camera/depth_registered_rectify_depth-15]: started with pid [9378]
process[camera/points_xyzrgb_hw_registered-16]: started with pid [9387]
process[camera/depth_registered_hw_metric_rect-17]: started with pid [9391]
process[camera/depth_registered_metric-18]: started with pid [9412]
process[camera/disparity_depth-19]: started with pid [9420]
process[camera/disparity_registered_sw-20]: started with pid [9424]
process[camera/disparity_registered_hw-21]: started with pid [9432]
process[camera_base_link-22]: started with pid [9443]
process[camera_base_link1-23]: started with pid [9445]
process[camera_base_link2-24]: started with pid [9463]
process[camera_base_link3-25]: started with pid [9471]
Warning: USB events thread - failed to set priority. This might cause loss of data...
[ INFO] [1499676943.699963616]: Number devices connected: 2
[ INFO] [1499676943.800937024]: 1. device on bus 001:17 is a SensorKinect (2ae) from PrimeSense (45e) with serial id '0'
[ INFO] [1499676943.801022556]: 2. device on bus 001:17 is a SensorV2 (2ae) from PrimeSense (45e) with serial id 'A00363920469048A'
[ INFO] [1499676943.802363361]: Searching for device with index = 1
[camera/camera_nodelet_manager-2] process has died [pid 9237, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-camera_nodelet_manager-2.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-camera_nodelet_manager-2*.log
[FATAL] [1499676955.288648044]: Failed to load nodelet '/camera/points_xyzrgb_hw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `camera_nodelet_manager'
[FATAL] [1499676955.289234539]: Failed to load nodelet '/camera/depth_registered_metric` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1499676955.289355207]: Failed to load nodelet '/camera/depth_registered_hw_metric_rect` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1499676955.289450084]: Failed to load nodelet '/camera/depth_registered_sw_metric_rect` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1499676955.289567134]: Failed to load nodelet '/camera/points_xyzrgb_sw_registered` of type `depth_image_proc/point_cloud_xyzrgb` to manager `camera_nodelet_manager'
[FATAL] [1499676955.289666340]: Failed to load nodelet '/camera/disparity_depth` of type `depth_image_proc/disparity` to manager `camera_nodelet_manager'
[FATAL] [1499676955.289845471]: Failed to load nodelet '/camera/disparity_registered_sw` of type `depth_image_proc/disparity` to manager `camera_nodelet_manager'
[FATAL] [1499676955.290129104]: Failed to load nodelet '/camera/depth_rectify_depth` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1499676955.290254835]: Failed to load nodelet '/camera/ir_rectify_ir` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1499676955.290365995]: Failed to load nodelet '/camera/driver` of type `openni_camera/driver` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291081899]: Failed to load nodelet '/camera/depth_registered_rectify_depth` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291270966]: Failed to load nodelet '/camera/rgb_rectify_color` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291462331]: Failed to load nodelet '/camera/depth_points` of type `depth_image_proc/point_cloud_xyz` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291763068]: Failed to load nodelet '/camera/depth_metric_rect` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291935707]: Failed to load nodelet '/camera/depth_metric` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291971826]: Failed to load nodelet '/camera/disparity_registered_hw` of type `depth_image_proc/disparity` to manager `camera_nodelet_manager'
[FATAL] [1499676955.291957955]: Failed to load nodelet '/camera/register_depth_rgb` of type `depth_image_proc/register` to manager `camera_nodelet_manager'
[camera/driver-3] process has died [pid 9238, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load openni_camera/driver camera_nodelet_manager --no-bond ir:=ir rgb:=rgb depth:=depth depth_registered:=depth_registered projector:=projector __name:=driver __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-driver-3.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-driver-3*.log
[camera/depth_metric-10] process has died [pid 9322, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/convert_metric camera_nodelet_manager --no-bond image_raw:=depth/image_raw image:=depth/image __name:=depth_metric __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-depth_metric-10.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-depth_metric-10*.log
[camera/ir_rectify_ir-7] process has died [pid 9267, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/rectify camera_nodelet_manager --no-bond image_mono:=ir/image_raw image_rect:=ir/image_rect_ir __name:=ir_rectify_ir __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-ir_rectify_ir-7.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-ir_rectify_ir-7*.log
[camera/depth_points-11] process has died [pid 9337, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyz camera_nodelet_manager --no-bond image_rect:=depth/image_rect_raw points:=depth/points __name:=depth_points __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-depth_points-11.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-depth_points-11*.log
[camera/depth_registered_sw_metric_rect-14] process has died [pid 9366, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/convert_metric camera_nodelet_manager --no-bond image_raw:=depth_registered/sw_registered/image_rect_raw image:=depth_registered/sw_registered/image_rect __name:=depth_registered_sw_metric_rect __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-depth_registered_sw_metric_rect-14.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-depth_registered_sw_metric_rect-14*.log
[... 10 lines cut ...]
[camera/register_depth_rgb-12] process has died [pid 9343, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/register camera_nodelet_manager --no-bond rgb/camera_info:=rgb/camera_info depth/camera_info:=depth/camera_info depth/image_rect:=depth/image_rect_raw depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw __name:=register_depth_rgb __log:=/home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-register_depth_rgb-12.log].
log file: /home/pdw/.ros/log/8c588874-654d-11e7-ab6f-00e0b412588a/camera-register_depth_rgb-12*.log
[... cut ...]
http://answers.ros.org/question/135266/problem-running-openni_launchlaunch-failed-to-load-nodelet-cameradriver-of-type-openni_cameradriver-could-not-find-library-corresponding-to-plugin/
http://answers.ros.org/question/54863/camera_nodelet_manager-1-process-has-died-error-with-kinect/

But it doesn't work.
I can run the standalone NIViewer with kinect. So I think the kinect driver should be OK on ubuntu 16.04 LTS.
Has anyone found the way to walk through this?
I think this issue may have sth. to do with this one:
https://github.com/ros-drivers/openni_camera/issues/18

130s, 2017-07-11:
Can you try #52 as mentioned in https://github.com/ros-drivers/openni_camera/issues/18#issuecomment-296549664 and see if that fixes your issue?

blueeaglex, 2017-07-11:
@130s  I tried. But I got compile error:
` In member function ‘void openni_camera::DriverNodelet::onInitImpl()’:
/home/pdw/dev/openni_catkin_ws/src/openni_camera/src/nodelets/driver.cpp:149:3: error: ‘LoggerPtr’ is not a member of ‘log4cxx’`

blueeaglex, 2017-07-11:
@130s 
Does  the branch 'rollback_usb' work with kinetic?

130s, 2017-07-11:
>  Does the branch 'rollback_usb' work with kinetic?

Only based on my assumption below with #52, yes it can/might work on ROS Kinetic:
- #52 is targeted to `indigo-devel` branch, which is the latest for this repo.
- [CI result](https://travis-ci.org/ros-drivers/openni_camera/builds/225092148) where Kinetic is tested passed without an issue.
  - This doesn't necessarily mean that every possible usecase is tested though.
  - Note that tests on CI run WITHOUT the actual hardware.

blueeaglex, 2017-07-12:
@130s 
I succeed compiling your commits below with #52.
And I download the openni_launch package.
This time I got a bit different things after I excuted openni.launch:
`[ INFO] [1499826307.214740439]: Number devices connected: 2
[ INFO] [1499826307.214866068]: 1. device on bus 001:11 is a SensorKinect (2ae) from PrimeSense (45e) with serial id ''
[ INFO] [1499826307.214929254]: 2. device on bus 001:11 is a Xbox NUI Camera (2ae) from Microsoft (45e) with serial id 'A00363920469048A'
[ INFO] [1499826307.216282417]: Searching for device with index = 1
nodelet: /usr/include/boost/smart_ptr/shared_ptr.hpp:641: typename boost::detail::sp_dereference<T>::type boost::shared_ptr<T>::operator*() const [with T = xn::NodeInfo; typename boost::detail::sp_dereference<T>::type = xn::NodeInfo&]: Assertion `px != 0' failed.
[camera/camera_nodelet_manager-1] process has died [pid 16461, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/pdw/.ros/log/3149da4e-66a8-11e7-8e89-00e0b412588a/camera-camera_nodelet_manager-1.log].
`

blueeaglex, 2017-07-12:
@130s 
I think I've got something.
It looks like the detectet serial id of SensorKinect is invalid.
It shouldn't be 0 or empty.

blueeaglex, 2017-07-13:
@130s  When I compile the code，I see lots of warning like this
[image]
Using deprecated functions. Should I install an old version of openni?

blueeaglex, 2017-07-13:
@130s I've already got the point.
The function [ OpenNIDriver::updateDeviceList()](https://github.com/ros-drivers/openni_camera/blob/indigo-devel/src/nodelets/driver.cpp#L223) returns two device item when I conect my kinect sensor to my pc. But as you can see,the first one's serial ID is invalid.
[image]
So I jump over the first one:

And the program just run normal now!
But it's not over. This is just a temp solution. I need time to locate the actual code where the bug comes from. Or maybe It is just because of the driver?
Do you have any suggestions?

blueeaglex, 2017-07-20:
This issue finally comes to an end. @fmnoori
All problems come from that I ever installed  http://github.com/avin2/SensorKinect.git
There was no need for this driver.
After I uninstalled the driver,there was no output of 'device on bus XXX is a SensorKinect' with a invalid serial id any more but only a normal 'Xbox NUI Camera' with an valid serial id.
Topic solved for now. :) 
But it is still hard to know what's going on with openni_camera. The standalone OpenNI was never  disturb by the incorrect driver I installed by openni_camrea was. @130s

TanayaBhave, 2018-08-01:
@blueeaglex How to uninstall the driver ? Could you please tell me the command