ros-drivers/openni_camera #55: Failed to load nodelet [/camera/rgb_rectify_mono] of type [image_proc/rectify] even after refreshing the cache: Failed to load library /home/ap4isr/kinetic_workspc/devel/lib//libimage_proc.

fmnoori, 2017-07-14:
.kinect1.0 for xbox360
Ubuntu16.04 LTS, ros-kinetic-desktop-full
I tried to excute openni.launch and get these errors: 
How to sort the problem?

ap4isr@stop-01:~$ roslaunch openni_launch openni.launch
... logging to /home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/roslaunch-stop-01-14616.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://stop-01:43551/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_hw/max_range: 4.0
 * /camera/disparity_registered_hw/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_registration: False
 * /camera/driver/device_id: #1
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /camera/
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_hw_metric_rect (nodelet/nodelet)
    depth_registered_metric (nodelet/nodelet)
    depth_registered_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_hw (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    driver (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_hw_registered (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)
  /
    camera_base_link (tf2_ros/static_transform_publisher)
    camera_base_link1 (tf2_ros/static_transform_publisher)
    camera_base_link2 (tf2_ros/static_transform_publisher)
    camera_base_link3 (tf2_ros/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[camera/camera_nodelet_manager-1]: started with pid [14634]
process[camera/driver-2]: started with pid [14635]
process[camera/rgb_debayer-3]: started with pid [14636]
process[camera/rgb_rectify_mono-4]: started with pid [14649]
[... 35 lines cut ...]
[ERROR] [1500052587.713067546]: The error before refreshing the cache was: Failed to load library /home/ap4isr/kinetic_workspc/devel/lib//libimage_proc.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.1: cannot open shared object file: No such file or directory)
[FATAL] [1500052587.713233598]: Failed to load nodelet '/camera/rgb_rectify_color` of type `image_proc/rectify` to manager `camera_nodelet_manager'
[camera/rgb_debayer-3] process has died [pid 14636, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/debayer camera_nodelet_manager --no-bond image_raw:=rgb/image_raw image_mono:=rgb/image_mono image_color:=rgb/image_color __name:=rgb_debayer __log:=/home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/camera-rgb_debayer-3.log].
log file: /home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/camera-rgb_debayer-3*.log
[ERROR] [1500052587.857666409]: Failed to load nodelet [/camera/depth_metric_rect] of type [depth_image_proc/convert_metric] even after refreshing the cache: Failed to load library /home/ap4isr/kinetic_workspc/devel/lib//libdepth_image_proc.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.1: cannot open shared object file: No such file or directory)
[ERROR] [1500052587.857718832]: The error before refreshing the cache was: Failed to load library /home/ap4isr/kinetic_workspc/devel/lib//libdepth_image_proc.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.1: cannot open shared object file: No such file or directory)
[FATAL] [1500052587.858070640]: Failed to load nodelet '/camera/depth_metric_rect` of type `depth_image_proc/convert_metric` to manager `camera_nodelet_manager'
[camera/rgb_rectify_color-5] process has died [pid 14658, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/rectify camera_nodelet_manager --no-bond image_mono:=rgb/image_color image_rect:=rgb/image_rect_color __name:=rgb_rectify_color __log:=/home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/camera-rgb_rectify_color-5.log].
log file: /home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/camera-rgb_rectify_color-5*.log
Warning: USB events thread - failed to set priority. This might cause loss of data...
Warning: USB events thread - failed to set priority. This might cause loss of data...
[ INFO] [1500052588.061274833]: Number devices connected: 1
[ INFO] [1500052588.061441965]: 1. device on bus 001:18 is a SensorKinect (2ae) from PrimeSense (45e) with serial id '0000000000000000'
[ INFO] [1500052588.063661098]: Searching for device with index = 1
[camera/depth_metric_rect-8] process has died [pid 14689, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load depth_image_proc/convert_metric camera_nodelet_manager --no-bond image_raw:=depth/image_rect_raw image:=depth/image_rect __name:=depth_metric_rect __log:=/home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/camera-depth_metric_rect-8.log].
log file: /home/ap4isr/.ros/log/28d7e128-68b8-11e7-9672-b8819811d451/camera-depth_metric_rect-8*.log
[ INFO] [1500052589.130208680]: Opened 'SensorKinect' on bus 1:18 with serial number '0000000000000000'
[ INFO] [1500052589.201266720]: rgb_frame_id = 'camera_rgb_optical_frame' 
[ INFO] [1500052589.201327489]: depth_frame_id = 'camera_depth_optical_frame' 
[ WARN] [1500052589.207456685]: Camera calibration file /home/ap4isr/.ros/camera_info/rgb_0000000000000000.yaml not found.
[ WARN] [1500052589.207565532]: Using default parameters for RGB camera calibration.
[ WARN] [1500052589.207637147]: Camera calibration file /home/ap4isr/.ros/camera_info/depth_0000000000000000.yaml not found.
[ WARN] [1500052589.207677358]: Using default parameters for IR camera calibration.
[ERROR] [1500052589.342204110]: Failed to load nodelet [/camera/depth_rectify_depth] of type [image_proc/rectify] even after refreshing the cache: Failed to load library /home/ap4isr/kinetic_workspc/devel/lib//libimage_proc.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libopencv_imgproc3.so.3.1: cannot open shared object file: No such file or directory)
[ERROR] [1500052589.342268729]: The error before refreshing the cache was: Failed to load library /home/ap4isr/kinetic_workspc/devel/lib//libimage_proc.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exceptio

130s, 2017-07-14:
@fmnoori Sorry to sound strict, but please go through [community etiquette](http://wiki.ros.org/Support#Etiquette). In particular, "Please don't send your question more than once" as you've already [reported the same issue](https://github.com/ros-drivers/openni_launch/issues/31#issuecomment-315417089). If I'm wrong then I'm sorry and please explain how different this is.

closed 2017-07-14 by 130s

fmnoori, 2017-07-14:
Sorry, my bad.  I will take care next time. :)