ros-drivers/openni_camera #63: Problem with IR image

130s, 2018-01-06:
[image] **Issue by [jokla](https://github.com/jokla)**
_Wednesday Sep 02, 2015 at 08:35 GMT_
_Originally opened as https://github.com/ros-drivers/openni_launch/issues/20_

----

Hi,

I'm trying to [calibrate the Kinect depth camera to the built-in RGB camera](http://wiki.ros.org/openni_launch/Tutorials/ExtrinsicCalibration). I calibrate the intrinsic parameters without any problem but I have a problem with the IR image:
running `rosrun image_view image_view image:=/camera/ir/image_raw` I just see a black image, while during the calibration with `rosrun camera_calibration cameracalibrator.py image:=/camera/ir/image_raw camera:=/camera/rgb --size 5x4 --square 0.0245` the IR image was displayed correctly.

For this reason I cannot compute the extrinsic parameters between the RGB and the IR camera using camera_pose_calibration node... 

With rqt_gui I can see this:
[image]

while running the camera_pose_calibration (the rgb images are coming from a bag file):
[image]

I am using Ubuntu 14.04 and ROS Indigo. I had to install the older version of driver manually:
 ( [libopenni-sensor-primesense-dev and libopenni-sensor-primesense0](https://launchpad.net/~v-launchpad-jochen-sprickerhof-de/+archive/ubuntu/pcl/+build/5252451)). I can see from rqt_graph that both node are subscribing to the same topic...

Any ideas?

Thank you!

BrainPotter, 2019-05-14:
Did you solve this problem already? I met the same error as you mentioned.

LexTheProgrammer, 2021-06-25:
Hi, same error here, did you find any solution?