ros-drivers/openni_camera #68: Melodic Release?

DLu, 2018-08-29:


k-okada, 2018-09-12:
@130s if you can not release this to newer rosdistro (melodic), can someone @wjwwood ? @dirk-thomas ? give me write access to this repository and https://github.com/ros-gbp/openni_camera-release, so that I can release as a ROS Orphaned Package Maintainer initiatives ?

130s, 2018-09-13:
- No update in the default branch since the last release to previous releases.
- There's no blocker for releasing `openni_camera` according to [repositories.ros.org](http://repositories.ros.org/status_page/blocked_releases_melodic.html?q=openni_camera&s=6&r=1).

With that, I just ran bloom. Then got a following error. Since this is easy to be fixed I'll create an immediate patch and re-run bloom soon.

```
$ bloom-release --rosdistro melodic --track melodic openni_camera
:
==> git-bloom-generate -y rosdebian --prefix release/melodic melodic -i 0 --os-name ubuntu
Generating source debs for the packages: ['openni_launch', 'openni_description', 'openni_camera']
Debian Incremental Version: 0
Debian Distributions: ['bionic']
Releasing for rosdistro: melodic

Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Could not resolve rosdep key 'kobuki_description'
Failed to resolve kobuki_description on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'kobuki_description', aborting.
kobuki_description is depended on by these packages: ['openni_description']
<== Failed
Could not resolve rosdep key 'turtlebot_description'
Failed to resolve turtlebot_description on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'turtlebot_description', aborting.
turtlebot_description is depended on by these packages: ['openni_description']
<== Failed
Some of the dependencies for packages in this repository could not be resolved by rosdep.
You can try to address the issues which appear above and try again if you wish.
```

closed 2018-09-13 by 130s

130s, 2018-09-13:
Release made https://github.com/ros/rosdistro/pull/19117