ros-drivers/openni_camera #77: Noetic release request

knorth55, 2022-04-08:
Thank you for this useful package.
Are there any plan to release this package in Noetic?

130s, 2022-04-12:
Thank you for the headsup. Release of 1.6.0 is made on openni2_camera side.
.deb generation is requested in https://github.com/ros/rosdistro/pull/32739
Closing for now.

130s, 2022-04-12:
I'm sorry https://github.com/ros-drivers/openni_camera/issues/77#issuecomment-1096551058 was about openni2. Re-opening.

@knorth55 If you have verified the functionality of the pkgs on this repo on ROS noetic, wolud you mind opening a PR, had there been any code change needed to get it working? I just made `ros1` branch off of `indigo-devel`, which I intend to use as the main development branch for ROS Melodic and Noetic.

130s, 2022-04-12:
@knorth55 On a separate note, I added a maintainer right on this repo. Thank you for stepping up! Feel free to add yourself in `package.xml`.

knorth55, 2022-04-12:
@130s thank you! I will check for noetic!

knorth55, 2022-04-13:
@130s can you change the default branch from `indigo-devel` to `ros1` and give access to `ros-gbp/openni_camera-release`?
https://github.com/ros-gbp/openni_camera-release

knorth55, 2022-04-13:
Thanks to #67, python3 is already supported.

130s, 2022-04-15:
Thank you!  Because of the confirmation you did, I don't feel a necessity to cut another version release on this repo. What's needed instead was to publish into a new ROS distro i.e. Noetic, which is requested https://github.com/ros/rosdistro/pull/32783 (using `bloom`).

> can you change the default branch from indigo-devel to ros1 and give access to ros-gbp/openni_camera-release? https://github.com/ros-gbp/openni_camera-release

Done and done.

closed 2022-04-15 by 130s

knorth55, 2022-04-15:
You released the current version in Noetic! Perfect! Thank you!!