ros-drivers/openni_camera #79: ROS Melodic + Ubuntu Mate 18.04 + Kinect V1 + Raspberry PI 4B

mateusguilhermedasilva, 2024-02-21:
Hello, did anyone manage to get it to work on ROS Melodic + Ubuntu Mate 18.04 + Kinect V1 + Raspberry PI 4B? When I run the command `roslauch openni_laucn openni.launch` I receive the message stating that no device was found or no device connected. However, the Kinect V1 is connected correctly.

dierfeichai2, 2024-12-27:
Hello, I have exactly the same issue as you, and my environment is  ROS Melodic + Ubuntu 18.04 LTS + Kinect V1. Have you resolved it?

mateusguilhermedasilva, 2024-12-27:
Hello, in my case I was only able to solve it by migrating to ROS Noetic, Ubuntu 20.04 and x86 platform. If you haven't done so yet, check the permissions of the USB port and make sure it has full permission